TJU-Aerial-Robotics/YOPO #2: CV_bridge ERROR: The ROS path is not included in Python Path!

Mioulo, 2024-10-30:
CV_bridge ERROR: The ROS path is not included in Python Path!
[ERROR] [1730218412.790221]: bad callback: <bound method YopoNet.callback_depth of <__main__.YopoNet object at 0x7ff2df5c3400>>
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "test_yopo_ros.py", line 157, in callback_depth
    depth_ = cv2.resize(depth_, (self.width, self.height), interpolation=cv2.INTER_NEAREST)
UnboundLocalError: local variable 'depth_' referenced before assignment

Lu-tju, 2024-10-30:
You can check in the terminal by entering the command:
```
python
from cv_bridge.boost.cv_bridge_boost import getCvType
```
you may get the output:
```
ImportError: /lib/libgdal.so.26: undefined symbol: TIFFReadRGBATileExt, version LIBTIFF_4.0
```
if so, you can find the missing lib by `locate libtiff.so.5` and then:
```
export LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libtiff.so.5
```
Besides, there may be other errors but the solution is similar:
```
example: undefined symbol: ffi_type_pointer, version LIBFFI_BASE_7.0
export LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libffi.so.7
```

closed 2024-11-08 by Mioulo

yuezhenyuan, 2025-04-15:
我也遇到了同样的问题，但当我在输入python和
from cv_bridge.boost.cv_bridge_boost import getCvType之后，输出Python 2.7.18 (default, Dec  9 2024, 19:35:20) 
[GCC 9.4.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> from cv_bridge.boost.cv_bridge_boost import getCvType
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: dynamic module does not define init function (initcv_bridge_boost)
>>> 
请问这种情况要怎么解决

Lu-tju, 2025-04-15:
> 我也遇到了同样的问题，但当我在输入python和 from cv_bridge.boost.cv_bridge_boost import getCvType之后，输出Python 2.7.18 (default, Dec 9 2024, 19:35:20) [GCC 9.4.0] on linux2 Type "help", "copyright", "credits" or "license" for more information.
> 
> > > > from cv_bridge.boost.cv_bridge_boost import getCvType
> > > > Traceback (most recent call last):
> > > > File "", line 1, in 
> > > > ImportError: dynamic module does not define init function (initcv_bridge_boost)
> 
> 请问这种情况要怎么解决

是python版本的问题吧。你是Ubuntu20.04吗？ubuntu20默认是python3不是python2

yuezhenyuan, 2025-04-15:
那我应该怎么修改呢

Lu-tju, 2025-04-15:
> 那我应该怎么修改呢

你是ubuntu20.04吗？是的话你尝试用python3取代python，再from cv_bridge.boost.cv_bridge_boost import getCvType

yuezhenyuan, 2025-04-17:
非常感谢作者，我刚刚在虚拟环境里面按你说的方法解决了问题。另外想问，作者会出实机教程吗

Lu-tju, 2025-04-17:
> 非常感谢作者，我刚刚在虚拟环境里面按你说的方法解决了问题。另外想问，作者会出实机教程吗
实机部署还好，因为是在微分平坦空间做的规划，也是用的高老师他们的仿真模型，和他们的控制器可以直接衔接，算法上不需要修改什么。实际上，实机主要的问题在Sim2Real，目前这个只有丛林环境，可以先选择类似的场景实验，后续我会丰富下场景

dynamic-tyrant, 2025-05-23:
CV_bridge ERROR: Possible solutions may be found at https://github.com/TJU-Aerial-Robotics/YOPO/issues/2
[ERROR] [1747972208.314207]: bad callback: <bound method YopoNet.callback_depth of <__main__.YopoNet object at 0x7f1d9294bd90>>
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "test_yopo_ros.py", line 164, in callback_depth
    if depth_.shape[0] != self.height or depth_.shape[1] != self.width:
UnboundLocalError: local variable 'depth_' referenced before assignment
我遇到了一个新问题，用之前的解决办法，并不能解决

ZhouZhiwei2019, 2025-06-05:
> CV_bridge ERROR: Possible solutions may be found at [#2](https://github.com/TJU-Aerial-Robotics/YOPO/issues/2) [ERROR] [1747972208.314207]: bad callback: <bound method YopoNet.callback_depth of [...]

同样的问题，请问找到解决方法了吗

Lu-tju, 2025-07-03:
每个人的环境会不一样不能一概而论，这个问题发生在python调用cv_bridge中，而在ros回调函数一旦出错会跳过此次回调，并不会打印错误信息，所以需要手动在终端执行cv_bridge查看具体报错信息解决：
```
> python
> from cv_bridge.boost.cv_bridge_boost import getCvType
```

或者在IDE中断点执行回调函数的cv_bridge部分查看具体报错：
```
depth = self.bridge.imgmsg_to_cv2(data, "32FC1")
```

Lu-tju, 2025-07-03:
已经添加备选方案代替cv_brigde:

```
try:
   depth = self.bridge.imgmsg_to_cv2(data, "32FC1")
except:
    assert data.encoding == "32FC1", f"Expected encoding '32FC1', got {data.encoding}"
    depth = np.frombuffer(data.data, dtype=np.float32).reshape(data.height, data.width)
```
可以更新至YOPO-Simple最新分支