TJU-Aerial-Robotics/YOPO #3: run_yopo.py 段错误 (核心已转储)

JinMohi, 2024-10-31:
Hello, Thank you so much for sharing this amazing work. Really grateful for this.
When I run the " python run_yopo.py --train=1",and I got the error:

/home/god/.conda/envs/yopo/lib/python3.8/site-packages/numpy/core/getlimits.py:499: UserWarning: The value of the smallest subnormal for <class 'numpy.float64'> type is zero.
  setattr(self, word, getattr(machar, word).flat[0])
/home/god/.conda/envs/yopo/lib/python3.8/site-packages/numpy/core/getlimits.py:89: UserWarning: The value of the smallest subnormal for <class 'numpy.float64'> type is zero.
  return self._float_to_str(self.smallest_subnormal)
/home/god/.conda/envs/yopo/lib/python3.8/site-packages/numpy/core/getlimits.py:499: UserWarning: The value of the smallest subnormal for <class 'numpy.float32'> type is zero.
  setattr(self, word, getattr(machar, word).flat[0])
/home/god/.conda/envs/yopo/lib/python3.8/site-packages/numpy/core/getlimits.py:89: UserWarning: The value of the smallest subnormal for <class 'numpy.float32'> type is zero.
  return self._float_to_str(self.smallest_subnormal)
段错误 (核心已转储)

Lu-tju, 2024-11-01:
Hello, I guess it's caused by the dataset. Have you collected the data before training? Check that the path of your dataset to be same as the configuration file `configs/vec_emv.yaml`:
```
dataset_path: "/run/yopo_sim/"
ply_path: "/run/yopo_sim/"
```
(The FLIGHTMARE_PATH is automatically added, for example, the absolute path will be: `FLIGHTMARE_PATH/run/yopo_sim
If this doesn’t work, you can use Python’s breakpoint debugging to locate the error in more detail (even though the error is in C++ part) to provide more information. Feel free to contact me anytime!

JinMohi, 2024-11-01:
I'm sure the path of my dataset is right.
Through debugging, I found that the issue seems to occur in line 56 of the code: 

train_env = QuadrotorEnv_v1(dump(cfg, Dumper=RoundTripDumper), False). 

Using gdb for debugging, It seems that the issue occurs in the generateMaps() function,but I don't know how to solve it.The gbd provided the following output:

[  #0  0x00007fff1c894b1b in std::vector<std::filesystem::__cxx11::path::_Cmpt, std::allocator<std::filesystem::__cxx11::path::_Cmpt> >::~vector() () from /home/god/jing/YOPO/flightlib/build/libflightlib.so
#1  0x00007fff1c894b39 in std::vector<std::filesystem::__cxx11::path::_Cmpt, std::allocator<std::filesystem::__cxx11::path::_Cmpt> >::~vector() () from /home/god/jing/YOPO/flightlib/build/libflightlib.so
#2  0x00007fff1c894bb9 in std::filesystem::__cxx11::path::~path() ()
   from /home/god/jing/YOPO/flightlib/build/libflightlib.so
#3  0x00007fff1c8ae929 in flightlib::VecEnv<flightlib::QuadrotorEnv>::generateMaps() ()
   from /home/god/jing/YOPO/flightlib/build/libflightlib.so
#4  0x00007fff1c8b2cf2 in flightlib::VecEnv<flightlib::QuadrotorEnv>::init() ()
   from /home/god/jing/YOPO/flightlib/build/libflightlib.so
#5  0x00007fff1c8b3bf8 in flightlib::VecEnv<flightlib::QuadrotorEnv>::VecEnv(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool) ()
   from /home/god/jing/YOPO/flightlib/build/libflightlib.so
#6  0x00007fff1ca2edac in ?? ()
   from /home/god/jing/YOPO/flightlib/build/flightgym.cpython-38-x86_64-linux-gnu.so
#7  0x00007fff1ca40423 in ?? ()
   from /home/god/jing/YOPO/flightlib/build/flightgym.cpython-38-x86_64-linux-gnu.so
#8  0x00000000004f5ad2 in cfunction_call_varargs (kwargs=<optimized out>, args=<optimized out>, 
    func=0x7ffff3f4bea0) at /usr/local/src/conda/python-3.8.20/Objects/call.c:743
#9  PyCFunction_Call (func=0x7ffff3f4bea0, args=<optimized out>, kwargs=<optimized out>)
    at /usr/local/src/conda/python-3.8.20/Objects/call.c:773
#10 0x00000000004e10cb in _PyObject_MakeTpCall (callable=0x7ffff3f4bea0, args=<optimized out>](url)

In addition, although I'm confident that my dataset path is correct, when I inspect the .tif files in my dataset, each file appears almost identical. I'm not sure if this is correct.

[image]
sorry,This is my first time using GitHub, and I'm not very familiar with some of the formatting.

Lu-tju, 2024-11-01:
The visualization of images is an issue with Ubuntu's image viewer. You can open it in PyCharm, where it will display correctly. Regarding the mapping issue, the dataset should looks like:
```
yopo_sim
├── 0
├── 1
├── pointcloud-0.ply 
└── pointcloud-1.ply 
```
I'm not quite sure for the other reasons, because it works fine on my computer,  you can try a small data set of mine:
https://drive.google.com/file/d/1-0feDabwLcc8T6Wb5vLIMzixdG8MoOMa/view?usp=drive_link

Lu-tju, 2024-11-01:
You can add my WeChat(15802288256) for a more timely reply and easier troubleshooting

closed 2024-11-07 by JinMohi

myFrise, 2024-12-08:
hey, I meet the same problem, how did you solve it?

Lu-tju, 2024-12-09:
This issue is fixed in the latest commit and you can pull the latest version

myFrise, 2024-12-09:
> > hey, I meet the same problem, how did you solve it?
> 
> Try to modify the `void VecEnv<EnvBase>::generateMaps()` in `vec_env.cpp` as following:
> 
> ```
> 	std::vector<std::string> ply_files;
> 	for (const auto& entry : boost::filesystem::directory_iterator(ply_path_)) {
>         if (boost::filesystem::is_regular_file(entry) && entry.path().extension() == ".ply") {
>             ply_files.push_back(entry.path().string());
>         }
>     }
> 
> 	// Sort according to the number of the filename.
> 	std::sort(ply_files.begin(), ply_files.end(), [this](const std::string& a, const std::string& b) {
> 		return extract_number(boost::filesystem::path(a).filename().string()) < extract_number(boost::filesystem::path(b).filename().string());
> 	});
> ```
> 
> It mainly modifies the `std::filesystem` to `boost::filesystem` because `std::filesystem` is added in C++17, but it looks C++11 you are using. I can't test this as I don't have this issue on multiple computers. If this works, please give me feedback so that I can fix it in the next version.

yes, it works, thank you!