TJU-Aerial-Robotics/YOPO #6: CV_bridge ERROR: Possible solutions may be found at https://github.com/TJU-Aerial-Robotics/YOPO/issues/2

fenglei920226, 2024-12-01:
When I run the command “python test_yopo_ros.py --trial=1 --epoch=0 --iter=0” in version 2.3, the following error occurs：

CV_bridge ERROR: Possible solutions may be found at https://github.com/TJU-Aerial-Robotics/YOPO/issues/2
[ERROR] [1733063332.522617]: bad callback: <bound method YopoNet.callback_depth of <__main__.YopoNet object at 0x7f286c1bd9a0>>
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "test_yopo_ros.py", line 145, in callback_depth
    if depth_.shape[0] != self.height or depth_.shape[1] != self.width:
UnboundLocalError: local variable 'depth_' referenced before assignment

fenglei920226, 2024-12-02:
I have resolved the issue. Thank you for your open-source work.

Lu-tju, 2024-12-02:
I am glad you solved it and thank you for your feedback

closed 2024-12-03 by Lu-tju