TJU-Aerial-Robotics/YOPO #8: the odometry and the odometry_ref in the test_yopo_ros.py

Zedzou, 2024-12-10:
hello, thank you for your code. when studying your method, I was confusion about the the odometry and the odometry_ref in the test_yopo_ros.py. Why do I need to enter two odometers.

Lu-tju, 2024-12-10:
Hi, `/juliett/ground_truth/odom` is the actual state given by the simulator, and  `/juliett/state_ref/odom` is the reference state given by the planner (reference state is also given to the controller at the same time with another topic). We use the reference state for planning instead of the actual state during prediction to ensure the continuity of the reference trajectory. Therefore we publish the reference state (pos, vel, acc) by the planner with the topic `/juliett/state_ref/odom` for the network. And at the beginning, there is no reference state, we use the actual state instead given by `/juliett/ground_truth/odom`.

Zedzou, 2024-12-10:
Hello, Thank you for your Patient explanation，If I try to deploy the YOPO on the UAV with the PX4 and the OriNX，using the vins-fusion to get the state，Do I just need to input vinsfusion's odometer into the juliett/groundtruth/odometer and input the depth to the depth image’s topic. And  I need to input anything to the /juliett/state-ref/odom?

---- Replied Message ----
| From | Lu ***@***.***> |
| Date | 12/10/2024 19:53 |
| To | TJU-Aerial-Robotics/YOPO ***@***.***> |
| Cc | Zedzou ***@***.***>,
Author ***@***.***> |
| Subject | Re: [TJU-Aerial-Robotics/YOPO] the odometry and the odometry_ref in the test_yopo_ros.py (Issue #8) |

Hi, /juliett/ground_truth/odom is the actual state given by the simulator, and /juliett/state_ref/odom is the reference state given by the planner (reference state is also given to the controller at the same time with another topic). We use the reference state for planning instead of the actual state during prediction to ensure the continuity of the reference trajectory. Therefore we publish the reference state (pos, vel, acc) by the planner with the topic /juliett/state_ref/odom for the network. And at the beginning, there is no reference state, we use the actual state instead given by /juliett/ground_truth/odom.

Lu-tju, 2024-12-10:
Hello, you do not need to manually publish the `/juliett/state-ref/odom`. It will be published by `yopo_planner_node` to `test_yopo.py` autonomously.

Zedzou, 2024-12-10:
Thank you very much. All I need to do is inputing the odometer and the depth image , and I run the test_yopo_ros.py, it will generate the trajectory, /juliett/state_ref/odom will be generated by the planner to ensure the continuity of the reference trajectory. right?

---- Replied Message ----
| From | Lu ***@***.***> |
| Date | 12/10/2024 20:29 |
| To | TJU-Aerial-Robotics/YOPO ***@***.***> |
| Cc | Zedzou ***@***.***>,
Author ***@***.***> |
| Subject | Re: [TJU-Aerial-Robotics/YOPO] the odometry and the odometry_ref in the test_yopo_ros.py (Issue #8) |

Hello, you do not need to manually publish the /juliett/state-ref/odom. It will be published by yopo_planner_node to test_yopo.py autonomously.

Lu-tju, 2024-12-10:
You are right. Please make sure that the topics of both `yopo_planner_node` and `test_yopo_ros.py`  are modified, the FOV is consistent with the training, and other parameters are modified correctly. (the `img_height` and `img_width` are the image size of the network instead of your camera, no need to modify them). You can check the topic connection with the `rqt_graph` and run the program to check the trajectory with UAV locked

Zedzou, 2024-12-10:
Ok. Thank you for your detailed guidance. I'll try it tomorrow，thank you very much.

---- Replied Message ----
| From | Lu ***@***.***> |
| Date | 12/10/2024 21:06 |
| To | TJU-Aerial-Robotics/YOPO ***@***.***> |
| Cc | Zedzou ***@***.***>,
Author ***@***.***> |
| Subject | Re: [TJU-Aerial-Robotics/YOPO] the odometry and the odometry_ref in the test_yopo_ros.py (Issue #8) |

You are right. Please make sure that the topics of both yopo_planner_node and test_yopo_ros.py are modified, the FOV is consistent with the training, and other parameters are modified correctly. (the img_height and img_width are the image size of the network instead of your camera, no need to modify them). You can check the topic connection with the rqt_graph and run the program to check the trajectory with UAV locked

Lu-tju, 2024-12-10:
You are welcome. By the way, the FOV I used and training is 90° (16 : 9), make sure your camera have the same settings

Zedzou, 2024-12-10:
Ok，did you use the d435i or d455 of?realsense

---- Replied Message ----
| From | Lu ***@***.***> |
| Date | 12/10/2024 21:10 |
| To | TJU-Aerial-Robotics/YOPO ***@***.***> |
| Cc | Zedzou ***@***.***>,
Author ***@***.***> |
| Subject | Re: [TJU-Aerial-Robotics/YOPO] the odometry and the odometry_ref in the test_yopo_ros.py (Issue #8) |

You are welcome. By the way, the FOV I used and training is 90° (16 : 9), make sure your camera have the same settings

Lu-tju, 2024-12-10:
I use d455, both of them have 90° (16 : 9) depth sensing

Zedzou, 2024-12-10:
Ok，Thank you for your patience.

---- Replied Message ----
| From | Lu ***@***.***> |
| Date | 12/10/2024 21:18 |
| To | TJU-Aerial-Robotics/YOPO ***@***.***> |
| Cc | Zedzou ***@***.***>,
Author ***@***.***> |
| Subject | Re: [TJU-Aerial-Robotics/YOPO] the odometry and the odometry_ref in the test_yopo_ros.py (Issue #8) |

I use d455, both of them have 90° (16 : 9) depth sensing

Zedzou, 2024-12-12:
Hello, Thank you for your Guidance. I have successfully deployed it on a real device.But from rviz,The predicted trajectory is far from the drone's location, and the trajectory keeps moving. Do you have any suggestions. My parameters are this:

```
test_yopo_ros.py:
def main():
    args = parser().parse_args()
    rsg_root = os.path.dirname(os.path.abspath(__file__))
    if args.use_tensorrt:
        weight = "yopo_trt.pth"
    else:
        weight = rsg_root + "/saved/YOPO_{}/Policy/epoch{}_iter{}.pth".format(args.trial, args.epoch, args.iter)

    settings = {'use_tensorrt': args.use_tensorrt,
                'network_frequency': 30,
                'img_height': 96,
                'img_width': 160,
                'goal': [20, 20, 2],           # the goal
                'env': '435',           # use Realsense D435 or Flightmare Simulator ('435' or 'flightmare')
                'pitch_angle_deg': -5,         # pitch of camera, ensure consistent with the simulator or your platform (no need to re-collect and re-train when modifying)
                'odom_topic': '/vrpn_client_node/uav1/pose',
                'depth_topic': '/camera/depth/image_rect_raw',
                'verbose': False,              # print the latency?
                'visualize': True              # visualize all predictions? set False in real flight
                }
    YopoNet(settings, weight)
```
```
int main(int argc, char** argv) {
	ros::init(argc, argv, "yopo_test");
	ros::NodeHandle nh;

	string cfg_path     = getenv("FLIGHTMARE_PATH") + std::string("/flightlib/configs/traj_opt.yaml");
	YAML::Node cfg_     = YAML::LoadFile(cfg_path);
	traj_time           = 2 * cfg_["radio_range"].as<double>() / cfg_["vel_max"].as<double>();
	int horizon_num     = cfg_["horizon_num"].as<int>();
	int vertical_num    = cfg_["vertical_num"].as<int>();
	int vel_num         = cfg_["vel_num"].as<int>();
	int radio_num       = cfg_["radio_num"].as<int>();
	double horizon_fov  = cfg_["horizon_camera_fov"].as<double>() * (horizon_num - 1) / horizon_num;
	double vertical_fov = cfg_["vertical_camera_fov"].as<double>() * (vertical_num - 1) / vertical_num;
	double vel_fov      = cfg_["vel_fov"].as<double>();
	double radio_range  = cfg_["radio_range"].as<double>();
	double vel_prefile  = cfg_["vel_prefile"].as<double>();

	getLatticeGuiding(lattice_nodes, horizon_num, vertical_num, radio_num, vel_num, horizon_fov, vertical_fov, radio_range, vel_fov, vel_prefile);
	traj_opt_bridge         = new TrajOptimizationBridge();
	traj_opt_bridge_for_vis = new TrajOptimizationBridge();

	lattice_trajs_visual_pub = nh.advertise<pcl::PointCloud<pcl::PointXYZI>>("/yopo_net/lattice_trajs_visual", 1);
	trajs_visual_pub         = nh.advertise<pcl::PointCloud<pcl::PointXYZI>>("/yopo_net/trajs_visual", 1);
	best_traj_visual_pub     = nh.advertise<pcl::PointCloud<pcl::PointXYZI>>("/yopo_net/best_traj_visual", 1);
	state_ref_pub            = nh.advertise<nav_msgs::Odometry>("/juliett/state_ref/odom", 10);

	// our PID Controller (realworld) & SO3 Controller (simulation)
	our_ctrl_pub = nh.advertise<quad_pos_ctrl::ctrl_ref>("/juliett_pos_ctrl_node/controller/ctrl_ref", 10);
	so3_ctrl_pub = nh.advertise<quadrotor_msgs::PositionCommand>("/so3_control/pos_cmd", 10);

	odom_sub      = nh.subscribe("/vrpn_client_node/uav1/odom", 1, yopo_net::odom_cb, ros::TransportHints().tcpNoDelay());
	yopo_best_sub = nh.subscribe("/yopo_net/pred_endstate", 1, yopo_net::yopo_cb, ros::TransportHints().tcpNoDelay());
	yopo_all_sub  = nh.subscribe("/yopo_net/pred_endstates", 1, trajs_vis_cb);
	goal_sub      = nh.subscribe("/yopo_net/goal", 1, goal_cb);

	ros::Timer ref_timer = nh.createTimer(ros::Duration(0.02), ref_pub_cb);
	std::cout << "YOPO Planner Node OK!" << std::endl;
	ros::spin();
```

Zedzou, 2024-12-12:
Thank you, I have solved. Change the _goal: [20, 20, 2]_ to _goal:[0, 0, 2],_ and it show _arrived_ in the terminal of yopo_planner_node, it works normal.

closed 2024-12-12 by Zedzou

Lu-tju, 2024-12-13:
Hello, I guess you just ran the program for testing without flying. If so, it is normal because the UAV does not follow the reference trajectory and the control error is getting larger and larger, which looks like the reference trajectory is getting farther away from the UAV. I suggest that do not modify the goal (keep [20,0,2]), but comment out the following in `yopo_planner_node_ros`:
```
ros::Timer ref_timer = nh.createTimer(ros::Duration(0.02), ref_pub_cb);
```
so that the planner will always plan from the actual position instead of the reference position (just for testing)