TJU-Aerial-Robotics/YOPO #10: 数据采集咨询

zhw970623, 2024-12-11:
请问大佬   除了(https://github.com/uzh-rpg/agile_autonomy)  公布的环境Flightmare Standalone 还有别的环境可以采集数据吗，这个数据对policy的影响怎么样

zhw970623, 2024-12-14:
> 你好： 仿真环境目前我仅使用[/uzh-rpg/agile_autonomy](https://github.com/uzh-rpg/agile_autonomy)开源环境，此外我还有一项开源环境 [SensorSimulator](https://github.com/TJU-Aerial-Robotics/SensorSimulator)，暂未添加到此仓库，你可以用来验证算法的泛化能力（我给出的训练权重仅在Flightmare中训练，在SensorSimulator也可以使用） 当然，添加更多RealWorld的数据肯定更有利于部署，因此，你可以用雷达来收集地图配合深度相机收集图像，混合到仿真数据集中一同训练，进一步增强可靠性（这很容易实现，因为我们的训练方法不需要专家标注，也不需要与环境进行交互）

非常感谢大佬解答，你们做的深度感觉已经超越ETH!!!   另外我在论文中看到在训练是需要获取ESDF信息，如果你们要做sim2real，能做到现实中零样本的部署吗（需要采集现实场景的ESDF信息吗）

Lu-tju, 2024-12-14:
零样本部署的话建议从低速开始尝试，我在[SensorSimulator](https://github.com/TJU-Aerial-Robotics/SensorSimulator)做过不同风格深度图的零样本迁移的测试，现实实验中稳妥起见我有采集真正的数据和仿真混合（测试场景是未知的，只是为了弥补下455和unity深度图的gap），要采集的数据和仿真训练一样：深度图、雷达点云、深度图对应的位置（无需收集ESDF，但需要运行雷达里程计如lio_sam以记录位置和地图。采集部分可简单参考run_data_collection_realworld.py）
此外，我在程序中也保留了[agile_autonomy](https://github.com/uzh-rpg/agile_autonomy)中双目匹配（SGM）的方式获取深度图，可以替换现在的仿真深度图

zhw970623, 2024-12-14:
谢谢解答~，非常感谢你们在未来扩展方面所作的工作。   另外，你们有实验过在森林场景采集ESDF数据指导规划器的训练，但是测试阶段换了场景（比如，废墟，雪地，室内）进行测试（无论是在现实还是仿真中），来看算法泛化性的表现吗？

Lu-tju, 2024-12-14:
深度图对场景的外貌具有鲁棒性（比如冬天雪地和夏天森林），但障碍的形状会有影响（比如楼宇、室内和树林会不同）
不过，我觉得泛化性的问题要通过扩展数据集来解决（正如传统的深度学习任务具有丰富的训练数据）而不仅是靠算法实现。因此，目前我正在进行一些数据扩展的工作，后续会将[SensorSimulator](https://github.com/TJU-Aerial-Robotics/SensorSimulator)替代该仓库的unity部分，通过生成随机障碍物来丰富训练场景，而不仅限于森林

closed 2024-12-20 by Lu-tju