TJU-Aerial-Robotics/YOPO #11: About the controler of the Realworld & PID Controller

Zedzou, 2024-12-13:
Hello. thank you for your code. I'm deploying your algorithm on a real machine, and want to know the __quad_pos_ctrl::ctrl_ref ctrl_msg__ in the _yopo_planner_node.cpp,_ is that your PID controler and open source? If I use the /mavros/setpoint_raw/global of the px4, can the same effect be achieved

Lu-tju, 2024-12-13:
the ctrl_msg `quad_pos_ctrl::ctrl_ref`  is sent to our PID position controller (which is not open source because it is not my work). Then, the controller publishes `/mavros/setpoint_raw/attitude` to px4. I took a look at the msg of `/mavros/setpoint_raw/global`, it looks like using the GPS coordinates (lat, long, alt)

Zedzou, 2024-12-13:
Thank you for your reply. That the msg  **/mavros/setpoint_raw/attitude,** I see that it not include the acceleration. the toipic **/mavros/setpoint_position/local** of the px4, It is the same type as the **ctrl_ref** message. Have you tried using PX4 control directly?

Lu-tju, 2024-12-13:
Yes, `/mavros/setpoint_raw/attitude` does not include the acceleration. The controller can be divided into position control and attitude control. We use our PID position controller (position, velocity, acceleration, and yaw, due to the differential flatness) and PX4's attitude controller (thrust and angular rate). 
I have not directly used PX4 control, but it is allowed to use PX4 for position control directly.

Zedzou, 2024-12-14:
Thank you. there is a [PID_Controller](https://github.com/TJU-Aerial-Robotics/PID_Controller), this is the one you use?

Lu-tju, 2024-12-14:
haha yes, I thought it was private...

Zedzou, 2024-12-14:
hahaha, thank you very much. I will do some experiments on the real robot. thank you.

closed 2024-12-14 by Zedzou

Lu-tju, 2024-12-14:
But in the latest high-speed experiment, I did not use this controller, the error is larger with the speed increasing.
Anyway, start slow.