TJU-Aerial-Robotics/YOPO #12: 怎么启动真飞呢？还需要修改代码吗？

2500490261, 2024-12-21:


Lu-tju, 2024-12-21:
1.  话题修改与目标点设置。修改推理节点` test_yopo_ros.py `的话题和你的VIO\深度相机对应：
```
'goal': [50, 0, 2],         #目标点
'env': '435',                 #深度图消息来源
'pitch_angle_deg':0, # 相机和机体的俯仰变换
'odom_topic': '/juliett/ground_truth/odom',# 你的视觉里程计
'depth_topic': '/depth_image',                        # 你的深度相机话题
```
修改控制发布节点`yopo_planner_node`的话题和你的VIO\控制器对应：
```
odom_sub = nh.subscribe("/juliett/ground_truth/odom", 1, yopo_net::odom_cb, ros::TransportHints().tcpNoDelay()); // 你的视觉里程计
so3_ctrl_pub = nh.advertise<quadrotor_msgs::PositionCommand>("/so3_control/pos_cmd", 10); // 你的无人机控制器
```
你可以通过ros_graph查看是否对应正确
2. 在traj_opt.yaml中修改飞行速度（首次飞行建议飞机处于锁定状态先检查轨迹是否正确），然后：
```
cd ~/YOPO/run
conda activate yopo
python test_yopo_ros.py --trial=1 --epoch=0 --iter=0
cd ~/YOPO/flightlib/build
./yopo_planner_node
```

2500490261, 2024-12-22:
> 1. 话题修改与目标点设置。修改推理节点`test_yopo_ros.py`的话题和你的VIO\深度相机对应：
>
> ```
> 'goal': [50, 0, 2],  #目标点
> 'env': '435',  #深度图消息来源
> 'pitch_angle_deg':0,  # 相机和机体的俯仰变换
> 'odom_topic': '/juliett/ground_truth/odom',  # [...]

> 1. 话题修改与目标点设置。修改推理节点`test_yopo_ros.py`的话题和你的VIO\深度相机对应：
>
> ```
> 'goal': [50, 0, 2],  #目标点
> 'env': '435',  #深度图消息来源
> 'pitch_angle_deg':0,  # 相机和机体的俯仰变换
> 'odom_topic': '/juliett/ground_truth/odom',  # [...]
感谢大佬

closed 2024-12-24 by Lu-tju