TJU-Aerial-Robotics/YOPO #13: Quantitative evaluation

Ecalpal, 2024-12-24:
Thanks for your great work!

The code runs well and is easy to read.

But I have a question, how can I obtain the quantitative results in Table 2 of the paper, such as Safety, Smoothness, and Traj. length?

Lu-tju, 2024-12-24:
Hi, in order to quantitative evaluation of safety, smoothness, and trajectory length, you can modify `simulator.launch`:
```
<arg name="init_x" value="-40.0"/>
<arg name="init_y" value="20.0"/>
<arg name="init_z" value="2.0"/> 
```
and run the following to evaluate the trajectories from x = [-40, 20] by default (just as a simple example, this repository has not organized the evaluation part). 
```
cd ~/YOPO/flightlib/build
./traj_eval_node
```
Besides, the depth images and camera intrinsic calibration are published at `/depth_image` and `/camera_info` when `flightros_node` starts, which can be used to evaluate the baselines.
For processing time, you can modify `verbose = True` in `test_yopo_ros.py` to print the latency in detail.
The script for success rate is not included in the repository, which can be easily implemented through a script.

Additionally, you may notice the results different from the reported (safer but less smooth) because the current project has been significantly modified, with different simulation environments and controllers from the original.

Ecalpal, 2024-12-24:
Thank you for your detailed response. 

I believe it will be helpful for anyone who wants to follow your work.