TJU-Aerial-Robotics/YOPO #17: What VIO method did you use in the real quadcopter.

Ecalpal, 2025-01-06:
Hi, 

I am trying to deploy your method on a real quadcopter. May I ask which VIO method you are using? Can you provide a deployment guide?

Ecalpal, 2025-01-06:
I have re-read your article and found that the VIO method you mentioned using is VINS-Fusion.

I would like to know what the term 'geometric controller' means in the phrase 'reference trajectory tracked by geometric controller' in the article.

Lu-tju, 2025-01-06:
Hi,
Yes, I have tried several VIOs, and VINS-Fusion is the best one I think. 
 'geometric controller' is a classical controller. You can refer to the paper `Geometric tracking control of a quadrotor UAV on SE(3)` and the controller in our repository, which is used in my paper.
However, the classical controller has large tracking errors when high-speed flight (30-50cm at 4-6m/s), and in my latest experiment it was deprecated. So I suggest starting slow for safety.

Ecalpal, 2025-01-06:
Thank you for your reply!

closed 2025-01-06 by Ecalpal