TJU-Aerial-Robotics/YOPO #20: How to control the max speed at inference time

ztqwerty, 2025-02-18:
Excellent work!

I have a question about speed control at inference time:

How to control the desired max speed, and possibly allow the drone to stop at inference time? 

I see there is a knob called "vel_max" in traj_opt.yaml, can it be used and adjusted at runtime dynamically?

Lu-tju, 2025-02-20:
Hi,
Yes, you can change the parameter `vel_max` to adjust the maximum desired speed without the need to retrain the network (the given weights are trained at 6m/s but work well at 0-6m/s). 
Additionally, I didn't understand what you meant by "possibly allow the drone to stop at inference time". During inference, the drone will execute the last trajectory. 
Furthermore, by "adjusted at runtime dynamically," do you mean adjusting different maximum speeds during a single flight? Theoretically, this is feasible, but I only use a fixed maximum speed during a single flight.

ztqwerty, 2025-02-24:
Thanks for the quick reply. What I am trying to achieve is to set the goal direction along with the desired speed, such that the drone will stop when the desired speed is 0, in this YOPO framework. Otherwise the drone will keep moving in the goal direction indefinitely. Is it something feasible?

Lu-tju, 2025-02-24:
The goal is a pre-set position in `test_yopo_ros.py`. The network inputs the goal direction, which is calculated online based on the current position of the drone and goal. When the drone approaches the pre-set goal position, it will switch to hover mode. If the drone overshoots or deviates from the goal position, it will return.
The network inputs a normalized unit direction vector without distance information,  which ensures consistent input for goals of different scales during training. So, I am afraid the way you mentioned could not be directly applied.

closed 2025-04-01 by Lu-tju