TJU-Aerial-Robotics/YOPO #21: “guidance” learning vs. reinforcement learning

bailvwangzi, 2025-02-25:
Didn't understand what the difference between “guidance” learning and reinforcement learning. The proposed guidance learning method directly computes gradients for the network based on the environment's reward. Compared to actor-critic based reinforcement learning, it doesn't have a critic network. Can i consider that it is also a kind of reinforcement learning? 

On the other hand, the paper EXPERIMENT baseline method is imitation learning method, have you tested reinforcement learning method?

Lu-tju, 2025-03-02:
In fact, I initially attempted reinforcement learning methods such as SAC, but I found that if the environment could directly provide the gradients required by the actor network, then the critic network could be discarded. This is the origin of proposed method, but I think it cannot be regarded as reinforcement learning, as it replaces the crucial design of reinforcement learning (such as the TD error, policy updates). 

Additionally, I initially tried reinforcement learning methods like SAC and PPO, but the results were not satisfactory. Traditional reinforcement learning is more limited to control problems with state inputs and is not suitable for perception tasks involving large backbone networks. I believe that the recent differentiable physics simulation is a significant improvement in reinforcement learning and has achieved impressive results, so it is worth a try.

bailvwangzi, 2025-03-03:
Thanks for your reply.  I once read a job article about differentiable physics simulation. _Back to Newton’s Laws- Learning Vision-based Agile Flight via Differentiable Physics_  . This article seems to be able to get good results through the simplest differentiable physics environment. What do you think of the comparison between your proposed method and the differentiable  physics simulation?

closed 2025-03-06 by bailvwangzi