TJU-Aerial-Robotics/YOPO #25: Question about preparation for real-world flight

wyj-bit, 2025-03-20:
I have recently read your papers and studied your work and I am really grateful for your code repository. However, I have some questions about actual flight.
 What kind of preparatory work is needed for actual flight? Can I directly use the `/juliett/ground_truth/odom` as VIO and  `/so3_control/pos_cmd` as controller in real world?
  Is it that all I need to do is to modify the topic publishing node of the depth image and launch the `so3_quadrotor_simulator simulator.launch` ?
Thank you very much >_<.

Lu-tju, 2025-03-28:
Hi,
to flight in real world, you only need to run the file `test_yopo_ros_new.py`. But there are something to modify:
```
settings = {'use_tensorrt': args.use_tensorrt,
                'img_height': 96,              # consistent with training
                'img_width': 160,
                'goal': [20, 20, 2],           # the actual goal
                'env': 'flightmare',           # '435'
                'pitch_angle_deg': -5,         # pitch of camera, ensure consistent with your platform 
                'odom_topic': '/odom',         # modify
                'depth_topic': '/depth_image'  # modify
}
```
It is recommended to replace the controller with your actual controller, since I am not using `so3_quadrotor_simulator ` in actual flight.
In addition, it is recommended to lock the drone for the first time to observe whether the trajectory is reasonable to avoid missing modifications.

wyj-bit, 2025-04-07:
Thanks for your reply!