TJU-Aerial-Robotics/YOPO #33: 5. Build the flightlib

Ecalpal, 2025-04-29:
When I tried to build flightlib on the Jesston platform, I encountered the following issue. Have you ever encountered the same problem?

perform `cmake ..`

output:
```
....
-- Getting Pybind11...
Cloning into 'pybind11-src'...
Already on 'master'
-- Pybind11 downloaded!
-- pybind11 v3.0.0 dev1
CMake Error at /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:146 (message):
  Could NOT find Python (missing: Python_LIBRARIES Python_INCLUDE_DIRS
  Development) (found suitable version "3.9.5", minimum required is "3.8")
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:393 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.16/Modules/FindPython/Support.cmake:2214 (find_package_handle_standard_args)
  /usr/share/cmake-3.16/Modules/FindPython.cmake:304 (include)
  externals/pybind11-src/tools/pybind11NewTools.cmake:54 (find_package)
  externals/pybind11-src/tools/pybind11Common.cmake:201 (include)
  externals/pybind11-src/CMakeLists.txt:238 (include)

-- Configuring incomplete, errors occurred!
```

My platform:
Orin NX 16GB
python3.8
cuda11.4
torch 1.11

Lu-tju, 2025-04-29:
Hi, you don't need to compile the project on the Jetson; you only need to run the test_yopo_ros.py file. So, you just need to set up the Python environment. The only thing to note is that you must manually install the PyTorch provided by NVIDIA.

Ecalpal, 2025-04-29:
Thank you for your quickly reply!

closed 2025-04-29 by Ecalpal