TJU-Aerial-Robotics/YOPO #38: Access to paper

zeeshanns, 2025-05-03:
Hi, this work looks really interesting! I'm currently unable to access the IEEE article. If it's not too much trouble, could you please share a copy with me via email? My address is zs9672907@gmail.com
Thank you!

Lu-tju, 2025-05-08:
Hi, I have sent it, please check.

Sologala, 2025-05-24:
@Lu-tju  Awesom and solid work you guys have done.  I  want a copy of pdf  to read about it  too .  Big Thanks for you.  🥰  dawsonwen123@gmail.com

Lu-tju, 2025-05-27:
> [@Lu-tju](https://github.com/Lu-tju) Awesom and solid work you guys have done. I want a copy of pdf to read about it too . Big Thanks for you. 🥰 [dawsonwen123@gmail.com](mailto:dawsonwen123@gmail.com)

Hi, @Sologala  I have sent it, please check.

closed 2025-09-08 by Lu-tju