TJU-Aerial-Robotics/YOPO #45: 运行报错问题

Jack2-mery, 2025-05-16:
原来我在跑这个代码是没有任何问题，但是最近就是突然重新build一下flightlib库后就一直出现这个报错Traceback (most recent call last):
  File "data_collection_simulation.py", line 8, in <module>
    from flightgym import QuadrotorEnv_v1
ImportError: /home/gcf/YOPO/flightlib/build/flightgym.cpython-38-x86_64-linux-gnu.so: undefined symbol: _ZN9flightlib6VecEnvINS_12QuadrotorEnvEE14savePointcloudEi

Lu-tju, 2025-05-17:
会不会是你在其他python虚拟环境activate时编译的flightlib?
尝试下激活YOPO虚拟环境，删掉build下除setup.py以外的其他文件，重新编译试试

Jack2-mery, 2025-05-17:
重新编译了好几次 然后重新从github上克隆从头开始编译还是会出现这个报错，就是在激活激活YOPO虚拟环境的基础上编译的

Jack2-mery, 2025-05-17:
pybind11的版本会影响这个问题吗？你们曾经遇到过吗

Lu-tju, 2025-05-17:
我遇到过的就是没有激活yopo环境时cmake会有问题，其余的情况没有遇到过，你`pip show flightgym`试一下看看输出是什么？

Jack2-mery, 2025-05-18:
pip show flightgym的命令应该输出是没啥问题的，但我现在有个疑惑就是我把这个代码重新跑一遍，也会出现上述问题感觉很奇怪🤓

[image]

Jack2-mery, 2025-05-19:
然后我通过ldd -r flightgym.cpython-38-x86_64-linux-gnu.so运行那个共享库，检测共享库的依赖关系也会出现一堆未被定义的错误

[image]

Lu-tju, 2025-05-19:
emmmm，我也不知道了ORZ。pip show flightgym看上去也很正常，之所以让你看看是因为我之前曾经弄了好几个版本，他索引错了，索引到另一个版本上去了导致运行出错。
这是我的ldd -r flightgym.cpython-38-x86_64-linux-gnu.so，看上去一切正常

[image]

Lu-tju, 2025-05-19:
问题和vec_env.cpp的：
```
template<typename EnvBase>
bool VecEnv<EnvBase>::savePointcloud(int ply_id) {
	if (!unity_ready_ || unity_bridge_ptr_ == nullptr)
		return false;
	Vector<3> min_corner = bounding_box_origin_ - 0.5 * bounding_box_;
	Vector<3> max_corner = bounding_box_origin_ + 0.5 * bounding_box_;
	unity_bridge_ptr_->generatePointcloud(min_corner, max_corner, ply_id, ply_path_, scene_id_, pointcloud_resolution_);
	return true;
}
```

以及对应.h的
```
bool savePointcloud(int ply_id);
```
部分有关，但我这里复现不出你的问题，所以只能你来试试看了，看上去比较正常不知道为什么没链接到

Jack2-mery, 2025-05-19:
> emmmm，我也不知道了ORZ。pip show flightgym看上去也很正常，之所以让你看看是因为我之前曾经弄了好几个版本，他索引错了，索引到另一个版本上去了导致运行出错。 这是我的ldd -r flightgym.cpython-38-x86_64-linux-gnu.so，看上去一切正常
> 
> [image]

嗯嗯 我在试试吧主要这个问题突然就出现了 上周五一切还挺正常的

Jack2-mery, 2025-05-19:
已经解决这个问题了，就是库路径的问题，一开始查找问题的思路错了，在运行python程序的时候手动输入 export LD_LIBRARY_PATH=/home/gcf/YOPO/flightlib/build:$LD_LIBRARY_PATH的指令或者通过gedit ~/.bashrc [在里面手动输入保存一下就可以，但确实要给博主的上网速度点个赞，回消息速度很快，而且非常有耐心（非常nice)  哈哈哈哈

[image]

closed 2025-05-19 by Jack2-mery

Lu-tju, 2025-05-19:
好的，感谢反馈！