TJU-Aerial-Robotics/YOPO #47: pointcloud saveing.......

dynamic-tyrant, 2025-05-21:
运行命令python data_collection_simulation.py，一直输出pointcloud saveing.....................................................................，请问怎么解决

Mioulo, 2025-05-21:
点云需要一定的时间保存，收集数据环境可能需要几个小时。unity很小是正常的，他与收集数据的图像尺寸相同，默认是160x90。如果需要查看可以直接用ros可视化图像。

Lu-tju, 2025-05-21:
pointcloud saveing....................................................................
这个过程比较久，慢的话可能需要1分钟左右，你是一直长期卡在这里进行不下去吗？

dynamic-tyrant, 2025-05-21:
> pointcloud saveing.................................................................... 这个过程比较久，慢的话可能需要1分钟左右，你是一直长期卡在这里进行不下去吗？

是的，跑了一个晚上都是输出..........这个

Lu-tju, 2025-05-22:
不应该这样的，我也不确定原因，因为我只是改动了一些接口，具体点云保存的实现都是封装进Unity里了，这来自于https://github.com/uzh-rpg/agile_autonomy，你有跑过他们的程序吗？
然后几个地方：
1、你的场景unity connect成功了吗？
2、Unity场景中的树已经正常生成了吗？
3、你有多试几次吗？
虽然unity窗口很小，但是你应该可以看清树是否已经生成

dynamic-tyrant, 2025-05-22:
> 不应该这样的，我也不确定具体的原因，因为我只是改动了一些接口，具体点云保存的实现都是封装进Unity里了，这来自于[agile_autonomy](https://github.com/uzh-rpg/agile_autonomy)，你有跑过他们的程序吗？ 然后几个地方： 1、你的场景unity connect成功了吗？（会有一个空旷的山地） 2、Unity场景中的树已经正常生成了吗？（虽然unity窗口很小，但是你应该可以看清树是否生成） 3、你有多试几次吗？（可能Unity会偶尔链接失败，但非常少见）

unity connect是连接成功了，场景中的树也能看到，也试过好几次了，但是依然处在pointcloud saveing ,请问正常情况下应该输出什么

Lu-tju, 2025-05-22:
data_collection_simulation正常会...........直到pointcloud.ply文件保存下来，通常我需要20-30s左右时间，然后就开始收集图像了（能看到一个进度条）
会和路径有关吗？比如你有中文路径？或者你手动检查下你设置的路径中有没有点云生成

Lu-tju, 2025-05-22:
好的，有发现具体的原因吗

closed 2025-05-22 by Lu-tju