TJU-Aerial-Robotics/YOPO #50: vec_env.yaml配置文件中的num_envs参数

JinxiCheng001, 2025-05-25:
您好！
请问vec_env.yaml配置文件中的num_envs参数是用来做什么呢？我看默认该参数为16，您在采集数据代码data_collection_simulation.py中将其设置为1（cfg["env"]["num_envs"] = 1），但是在yopo_algorithm.py中将batch_size设置与其相等（等于16，# batch size == num of env），这是为什么呢？
另外它和您在data_collection_simulation.py中设置的参数num_env有什么区别呢？我注意到您设置num_env=10的目的是为了在10个环境中采集数据（parser.add_argument("--num_env", type=int, default=10, help="num of env to change")）。
期待您的回复！谢谢！

[image]

[image]

Lu-tju, 2025-06-20:
> 您好！ 感谢您拨冗回复！我在进行测试的时候发现无人机的画面视角总是第一人称视角，如下图： [image] 我希望改为第三人称视角，以便能在看到无人机机体的同时获得更大更广阔的视野，请问我需要怎么修改呢？ （我觉得这个视角设置应该在配置文件/YOPO/flightlib/configs/quadrotor_ros.yaml中，我尝试对rgb_camera_left的t_BC和r_BC进行了修改以调整相机相对无人机的相对位置和姿态，并且调大width和height以获得更清晰的画面，但我仍然无法看到无人机机体。）

你好！
1、你给的图像是机载相机拍摄的画面，为了看到无人机自身不建议修改相机图像，这会影响飞机和图像的位置进而影响规划。你可以把图像设的大一点（640*360）这样flightmare的Unity界面就会增大，从Unity看就好了，但是Flightmare-Unity在我这里设大分辨率后渲染的比较慢，所以我默认给的很小。
2、如果要看飞机自身还需要修改飞机的大小，为了防止相机拍摄到飞机机体我把飞机大小设的很小， 你可以在`quadrotor_env.cpp`36行修改：`Vector<3> quad_size(0.01, 0.01, 0.01);`
3、此外，我最新提交了一个新的分支（默认分支）相比于之前的版本简化了非常多，速度也快很多，你可以看最新的代码（但删除了Unity部分，你如果需要可视化的话那就不要切换了）

Lu-tju, 2025-06-20:
只修改flightmare的unity界面分辨率而不修改相机图像分辨率吗: 不是的，相机和unity界面分辨率是相等的，我的意思是你不要改rgb_camera_left的t_BC和r_BC，改分辨率就行了。

新分支的可视化是有的，但是没有飞机自身的可视化，你只能在RVIZ看而没有Unity的RGB。在我这里新分支的仿真效率比旧分支快100多倍，硬件要求也更低，推荐使用新分支

JinxiCheng001, 2025-06-20:
好的，谢谢你！

JinxiCheng001, 2025-08-07:
您好！我注意到您在YOPO/config/config.py中对sgm_time进行了定义，sgm_time=2*radio_range/vel_max_train≈1.67s，这个时间表示对未来1.67秒的轨迹进行规划，请问这个时间是怎么确定的呢？更改这个时间对无人机规划会有影响吗？另外我还有一个疑问，就是在训练过程中分别有3个cost，分别是guidance_cost、safety_cost和smooth_cost，分别负责引导无人机飞向目标点、对障碍物进行避障和提高轨迹平滑度，其中我理解中guidance_cost与safety_cost理论上会发生冲突，因为guidance_cost趋向于让轨迹贴近无人机当前位置与目标点位置连线，而safety_loss会让轨迹为躲避障碍物在中间过程发生弯曲，请问你是如何平衡好这3个cost的权重的关系的呢？期待您的回复！谢谢！

Lu-tju, 2025-08-07:
你好！
1、这个时间是我按照经验设定的，这样能使得实际飞行最大速度尽量接近所设置的速度，也就是当轨迹长度=2*radio_range，在这个时长下速度能近似vel_max_train，但这只是一个经验值，没有什么具体的保证，你也可以进行修改，但是必须在训练和测试时保持一致。
2、你说的很对，即便是在传统方法中也存在不同代价矛盾的问题，一组参数难以适应不同速度、不同场景，在我最新的代码中我添加了根据速度等比缩放权重的部分，这样更有利于调参，如果你用的是旧代码可以更新下。关于具体怎么权衡，就是通过调参观察效果来做的

JinxiCheng001, 2025-08-21:
好的！谢谢您！请问您在真机实验过程中是否尝试过动态场景呢，YOPO对于动态场景的适应性如何呢？在论文中我看到YOPO是实时规划的，理论上应该具备动态场景自动避障的功能