TJU-Aerial-Robotics/YOPO #51: 关于多无人机

hlyjsd, 2025-05-27:
你好，我想请问如果添加了多架无人机，如何获取每一架无人机的准确位置和速度呢？
我在代码中没有找到发布ground_truth/odom的位置

Lu-tju, 2025-05-27:
我通过UAV Simulator发布的ground_truth/odom位置，并不在这个仓库中，你可以参考：
https://github.com/TJU-Aerial-Robotics/UAV_Simulator
或者一个简单的多机分支：
https://github.com/TJU-Aerial-Robotics/UAV_Simulator/tree/muti-uav-simulator
但是当前我的flightmare的仓库中并没有留多机的接口，需要你自己添加下

hlyjsd, 2025-05-29:
感谢您的回答！
我还有个问题，如果增大无人机尺寸再进行测试，代码运行速度会变慢很多，增加的耗时主要是在unity_brifge.cpp中handleOutput()，但保持无人机尺寸不变而增加多架无人机的话并不会明显增加这个耗时。
请问您知道这个问题的原因吗？我该怎么修改呢？

hlyjsd, 2025-05-29:
我现在不太清楚是因为渲染而导致的耗时还是碰撞检测而导致的耗时

Lu-tju, 2025-05-29:
你说的耗时就是渲染导致的，这个已经打包好Unity可执行文件了并不是很好修改，之前我有尝试黑屏可视化界面来减少渲染，但影响不大。
但是你现象跟我恰恰相反，修改无人机尺寸不会影响渲染速度，增加多无人机会影响（需要渲染更多传感器观测），我不太确认你说的这个现象的原因

99cipyl, 2025-09-16:
@Lu-tju 您好，不太理解您说的意思。您的意思是如果要在flightmare中添加多无人机的话必须通过https://github.com/TJU-Aerial-Robotics/UAV_Simulator才能将flightmare中多无人机的odom都发布出来嘛？flightmare和UAV_Simulator关系是什么呢

99cipyl, 2025-09-16:
> @99cipyl <https://github.com/99cipyl>
> 你好，楼主看的我的代码那会还是旧的版本（main分支），在最新的默认分支已经弃用了flightmare，以提升实时性并简化环境依赖。
> 对于你的提问：UAV_Simulator是仿真无人机的运动学和动力学，输入控制量、输出状态odom
> flightmare是仿真环境和渲染观测，给定odom、输出对应状态的图像。
>
> 因此Odom不是flightmare发布的而是UAV_Simulator发布的，如果添加多个飞机，首先在uav_simulator中添加多个飞机（不需要改很多程序，修改launch文件即可）然后还需要修改flightmare以支持多个飞机的图像渲染

非常感谢您的回复，就比如您在yopov2 track中说的，被跟踪的无人机是基于fast
planner，那我如果还要把这一块放到flightmare中的话就需要按您说的先去uav_simulator中添加多个飞机然后再修改flightmare么～

99cipyl, 2025-09-17:
> @99cipyl <https://github.com/99cipyl>
> 是的，不是很复杂，uav_simulator直接launch文件起多个飞机岔开下话题就行，它接收的控制指令和fast
> planner是一样的，flightmare参考我node中添加无人机的方式添加就行

好的 非常感谢！