TJU-Aerial-Robotics/YOPO #56: hover problem

Platanus86, 2025-07-04:
Excellent work! I followed your guidance and run test program successfully, but I noticed that once uav arrive at goal ,it hover with a constant height, how can I change this. For example,  if I set a 3D goal point(in test, there is only 2D goal), uav can can stop near this point. Thank you!

Lu-tju, 2025-07-04:
Hi, you can directly set 3D target points in the `test_yopo_ros.py`. The pre-trained model supports goals at different heights (except for purely vertical goals, as the camera faces forward). 

1. you can set directly
```
 settings = {'use_tensorrt': args.use_tensorrt,
                   'goal': [50, 0, 2],      # 3D goal
                  ......}
```
2. you can modify the height of goal clicked in RViz within the ROS callback.
```
    def callback_set_goal(self, data):
        self.goal = np.asarray([data.pose.position.x, data.pose.position.y, 2])
```
Finally, since it's forest environment, I fix the heights to 2.

Lu-tju, 2025-07-05:
In the `config`, the `pitch` determines the angle of the simulated camera and the drone.
In `test_yopo_ros.py`, the `settings` defines the angle that the planner assumes between the drone and the camera.

Please ensure that the pitch in `settings` matches the one in `config` when running in simulation,
and corresponds to your actual drone's hardware when running in real flight.

Platanus86, 2025-07-05:
Got it, thank you so much!

closed 2025-07-07 by Lu-tju