TJU-Aerial-Robotics/YOPO #62: 运行界面样子问题

Guo466, 2025-08-17:
我在test_yopo_ros.py第82行修改为使用cpu，但是在可视化界面并没有显示出完整的样子。

[image]

Lu-tju, 2025-08-24:
这个和test_yopo_ros.py没有关系，你的Simulator有正确编译和启动吗？你的虚拟机中可以使用CUDA吗？看上去你并没有得到深度图，时仿真的原因，很可能你的CUDA并没有正确配置

Keanu-Re, 2025-09-09:
@Lu-tju 请问仿真还需要额外下载依赖软件吗？

Lu-tju, 2025-09-10:
@Keanu-Re 我也是在wsl中使用的，如果没有图像、点云，就Simulator的原因，他依赖CUDA架构自动检测，请确认：
1、你在编译时有没有错误？如果CUDA架构自动检测失败，你的显卡是什么？我在RTX 20、30系列都可以，50系列检测不到但会报错，能明显看到
```
cd Simulator
catkin_make
```
2、有没有正常启动？
```
cd Simulator
source devel/setup.bash
rosrun sensor_simulator sensor_simulator_cuda
```
3、rviz -d yopo.rviz 开启位置正确